ARM-software/vscode-arm-debugger #1: Flash tasks fails with space in image path

franco-benedetti-symbiotech, 2024-02-08:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11 23H2
- VS Code Version: 1.86.1
- Extension Version: 0.4.0
- Target Device: STM32F103RB (on a NUCLEO-F103RB)
- Other extensions you installed: all extensions coming with Arm Keil Studio Pack (pre release version)
- When trying to flash the image to the device, it fails with ERROR(CLI37).
I used the blinky example for that board, but it's the same with other projects and boards.

I think that it's caused by the user directory containing a space when used for the --image option. 
However the path for the pdsc option looks like it's properly surrounded by quotes.
This is the log from the task:
``` powershell
Executing task: arm-debugger.flash: Flash Device 

armdbg.exe --program-only --target="066CFF484971754867182016" --image=c:\Users\Franco Benedetti\Documents\Progetti\project1\MainDevice\FW\test\arm\cmsis-rtos-blinky-c963c1e7\out\RTX_Blinky\STM32F103_Flash\RTX_Blinky.axf --connect-hold-reset=true --probe="ST-Link" --cmsis-pdsc="c:\Users\Franco Benedetti\AppData\Local\arm\packs\Keil\STM32F1xx_DFP\2.4.1\Keil.STM32F1xx_DFP.pdsc" --cmsis-device-name="STM32F103RB"

ERROR(CLI37): Unrecognised option: Benedetti\AppData\Local\arm\packs\Keil\STM32F1xx_DFP\2.4.1\Keil.STM32F1xx_DFP.pdsc

Flash failed
```

If I take the command and I execute directly in the shell, adding the quotes around the path of the image, it works properly:

``` powershell
PS C:\Users\Franco Benedetti\Documents\Progetti\project1\MainDevice\FW\test\arm\cmsis-rtos-blinky-c963c1e7> armdbg.exe --program-only --target="066CFF484971754867182016" --image="c:\Users\Franco Benedetti\Documents\Progetti\project1\MainDevice\FW\test\arm\cmsis-rtos-blinky-c963c1e7\out\RTX_Blinky\STM32F103_Flash\RTX_Blinky.axf" --connect-hold-reset=true --probe="ST-Link" --cmsis-pdsc="c:\Users\Franco Benedetti\AppData\Local\arm\packs\Keil\STM32F1xx_DFP\2.4.1\Keil.STM32F1xx_DFP.pdsc" --cmsis-device-name="STM32F103RB"  
[server] Starting debug server
[server] Waiting for debug server to start accepting connections
[server] Debug server started successfully

Stopping running target Cortex-M3 on connection
Connected to running target Cortex-M3
Execution stopped in Handler mode at 0xFFFFFFFE
0xFFFFFFFE   MOVS     r0,r0
Mapping segment 0x08000000 ~ 0x08001D20 (size 0x1D20)
Starting flash programming operation: Programming 7456 bytes to 0x08000000
Flash Progress: 0% (Erasing)
Flash Progress: 40% (Erased 8192 bytes)
Flash Progress: 40% (Erasing completed)
Flash Progress: 40% (Writing)
Flash Progress: 85% (Wrote 7456 bytes)
Flash Progress: 85% (Writing completed)
Flash Progress: 85% (Verifying)
Flash Progress: 95% (Verifying completed)
Flash Progress: 100% (Programming completed)
Flashing complete
Target has been reset
Disconnected from stopped target Cortex-M3
[server] Debug server shutdown application
[server] Debug server shut down attempted
[server] Done
```

I tried to add the single quote around the `${command:arm-debugger.getApplicationFile}` variable in the tasks.json, but that didn't work either.

jreineckearm, 2024-02-16:
Thank you, @franco-benedetti-symbiotech , for raising this issue!

I think you are right with your suspicion that this is related to the missing quotes and the space in the path. We are looking into this issue and will get back to you once we have a potential fix for the problem.

jreineckearm, 2024-03-01:
@franco-benedetti-symbiotech , we just released Arm Debugger v1.0.0 which should fix this issue.
Please let us know if you still see the problem.

franco-benedetti-symbiotech, 2024-03-05:
@jreineckearm , it's working now! Thanks!

thegecko, 2024-03-15:
Closing as resolved

closed 2024-03-15 by thegecko