ARM-software/vscode-arm-debugger #2: Unable to debug with Arm Debugger

gauthiergodart, 2024-02-09:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 10
- VS Code Version: 1.86.1
- Extension Version: 0.5.1
- Other extensions you installed (and if the issue persists after disabling them): Keil Studio extensions pack

When trying to debug with the new Arm Debugger extension instead of the previous Arm Embedded Debugger, I'm stuck with the following error, and the error message does not help me to understand what is wrong:
```shell
Using the LLVM DWARF Parser for this file ***.elf
ERROR(CLI237-TAD12-NAL52): 
! An error occurred processing MSDAP message
! Unable to create breakpoint at 0x000007BC
! Cannot attain state requested.
Disconnected from stopped target Cortex-M0
[server] Debug server shutdown application
[server] Debug server shut down attempted
[server] Done
ERROR(CLI237-COR160): 
! An error occurred processing MSDAP message
! Operation on closed connection
```
There is mention of some breakpoint, but I did not set any of it.

I'm using a private DFP (not publicly released yet). I have no problem using the previous extension with the exact same project and environment.

Could you give me input/some hints about these error messages?

jreineckearm, 2024-02-16:
@gauthiergodart , thanks for raising this issue!

Is it possible to share more information about the device you are using?

This message could point to not having full debug access to the device when trying to set the breakpoint for running to `debugFrom` (by default `main` of the application). Maybe it is in a secured bootloader at that point?

I appreciate if you cannot disclose more details. Since you work on a DFP, it may be worth to explore https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/coresight_setup.html and to check if the CMSIS Pack description can help with overcoming this situation.

gauthiergodart, 2024-02-22:
Hi @jreineckearm 
We had to switch to the previous Arm Embedded Debugger extension, which does not show this behavior. But I won't miss getting back to you when I'll find some time to try again the Arm Debugger extension.
I think that if the PDSC was at fault, we would have already noticed it with the other tools used (uVision, VS Code with Arm Embedded Debugger extension...), but I'll investigate anyway.