ARM-software/vscode-arm-debugger #4: Arm Debugger freezes

Aptahar, 2024-05-03:
Type: Bug Report

**Describe the bug**
- OS and Version: Ubuntu 23.10 | Ubuntu 24.04
- VS Code Version: 1.89.0
- Extension Version: 1.2.1
- Target Device: GD32F350RB
- Other extensions you installed (and if the issue persists after disabling them): Keil Pack Extensions

**To Reproduce**
Steps to reproduce the behavior:
1. Debug starts and work normaly
2. At some point when do 'Step Into' command, Debugger freezes with message in DEBUG CONSOLE (see Screenshot1) 
3. After that Target stay in Run mode and doesn't respond to any buttons: Pause, Restart, Stop
4. Restart button clear DEBUG CONSOLE
5. After some time later (few seconds) button Stop can stop Debugging, but not every time, some times only restart VScode is the solution.
6. Button Pause cause notification like on Screenshot2

**Code samples and logs**
Where are any logs can be finded?

tasks.json
```
{
    "version": "2.0.0",
    "tasks": [
        {
            "type": "arm-debugger.flash",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "program": "${command:arm-debugger.getApplicationFile}",
            "cmsisPack": "${command:device-manager.getDevicePack}",
            "deviceName": "${command:cmsis-csolution.getDeviceName}",
            "processorName": "${command:cmsis-csolution.getProcessorName}",
            "problemMatcher": [],
            "label": "Flash Device"
        },
        {
            "type": "shell",
            "label": "generate fw",
            "command": "./utils/generate_firmware_VScode.sh",
            "problemMatcher": [],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        }
    ]
}
```

**Screenshots**
Screenshot1:
[image]
Screenshot2:
[image]

Aptahar, 2024-05-03:
After some restart VScode have this message:  
[image]

Aptahar, 2024-05-06:
debugger fails in irq handlers

thegecko, 2024-05-06:
This popup message occurs when an existing armdbg process exists, often if it hasn't been terminated correctly. We are investigating this. Killing any armdbg/java processes (or rebooting) will fix this issue on the meantime.