ARM-software/vscode-arm-debugger #8: Debugger does not update content of variables

FerdinandEn, 2024-05-17:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows & Ubuntu
- VS Code Version: 1.89.1
- Extension Version: 1.1.2
- Target Device: STM32F756ZGT (Nucleo Board)
-  I have deleted all extensions and Keil Studio pack new installed

It is a simple project - blinking a led

````
#include "main.h"

volatile uint32_t count = 1;
volatile uint32_t count2 = 5;

/**
  * @brief Entry point of user program
  */
void mainThread(void)
{
  for(;;)
  {
    HAL_GPIO_TogglePin(redLed_GPIO_Port, redLed_Pin);
    count++;
    count2 = count2 + count;
    HAL_Delay(1000);
  }
}
````
I can compile, flash and start the debugger.

I can modify the code and the behavior change (blink frequency)

The global variables are modified in super loop.

I can see the content of the variables in Globals and I can add as watch.

If the programm is running the shown content of the variables does not change in both views (debug stopped) - blinking occurs.

Same behavior in Memory inspector.

If I change the initial value, I can see the change.

If I control the program in Dissambly and register view. I can see, that variable is modified as expected.

**To Reproduce**

see example code

FerdinandEn, 2024-05-17:
[001-Blinky.zip](https://github.com/ARM-software/vscode-arm-debugger/files/15347888/001-Blinky.zip)

FerdinandEn, 2024-05-22:
In a few more experiments, I found that the problem occurs when debugging a program that was converted from a uVision project. I created a project for the STM32H753ZGT directly and as a converted project. The debugging in the directly created program works, the converted one does not.

jkrech, 2024-06-06:
Maybe a stupid question: Do you load the ELF file build by uVision or the one built by CMSIS-Toolbox?
The difference is that uVision requires DWARF4 debug information whereas the Arm Debugger in VSCode requires DWARF5.

FerdinandEn, 2024-06-06:
@jkrech I translated it in Keil Studio and deleted the objects directory from uVision to be on the safe side. I used the converted program code. Where do I have to enter that I want DWARF5 debug information?

jkrech, 2024-06-06:
Sorry, the default in AC6 when using -g is still DWARF4, therefore it should not make a difference even though in uVision we explicitly force -gdwarf-4. Therefore this cannot be the root cause.

FerdinandEn, 2024-06-10:
I am using a `STM32F756ZGT` (Nucleo Board). Originally the project was created for `STM32F746ZGT` . The difference is only the Crypto/Hash-Unit. I tried both boards no difference.

I also have the same problem with the STM32H753ZI. However, only when I use a converted project. As soon as I create a project directly with Keil Studio, the variable view of the debugger works.

jreineckearm, 2024-06-11:
Forgot to mention: when I say `a uVision build`, I mean opening the `uvprojx` in uVision.

jreineckearm, 2024-06-11:
I just exported the uVision example to csolution with the latest MDK 5.40 version. The problem is still present. But the code sized difference improved. Now ~0.2KBytes.

@FerdinandEn , this makes me wonder how you did convert the uVision project to a csolution? Was it the export from uVision? Which MDK 5 version, if so? Or did you use the conversion in the CMSIS Solution extension?

FerdinandEn, 2024-06-11:
@jreineckearm In most cases, I carry out the conversions from within Keil Studio. I select the `uvprojx` file with the right mouse button and choose `Convert uvproj to csolution`. Due to the regular updates, I assume that I am using the most up-to-date conversion program.

MPU -> I have tested whether the variables are updated when I deactivate the MPU. The view then also works for me.

The MPU setting shown is suggested by ST. It has the effect that access is prohibited in the external memory area for memory and peripherals. This prevents incorrect cache accesses. However, activating the MPU also means that all memory areas, including the internal ones, can only be accessed in privileged mode. 

However, I also have programs where the display of variables works even if the MPU is active.

jreineckearm, 2024-06-11:
Thanks, @FerdinandEn , for the additional information! This is very useful. The required privileged access is actually a good hint to follow up on the debugger side.
We'll further investigate and get back to you.

FerdinandEn, 2024-10-16:
The problem still exists. If you select the MPU settings suggested by ST for the STM32F756ZGT - Background Region active, i.e. only privileged access possible then the variables do not update. 

I have just set up a new project with Keil Studio directly (Keil::STM32F7xx_DFP@3.0.0) 

The problem is still the same. If I deactivate the MPU setting, the debugger works. Debugging in Keil Studio remains difficult for the STM32F7. 

Can you tell me what the status of the troubleshooting is?

tomgolubev37, 2025-02-11:
Hi @FerdinandEn, 

Silly question, but the count variables are not used, so likely are "optimized out"  can you make count2 or count1 be used in the program, such as setting that t

count2 = 1000 + count;

Also, can play around with optimization levels, but I would not expect to be able to breakpoint on that line, or see count2 as a variable if I pause the debugger.

Single stepping can also expose which lines are getting executed.

--Tom

FerdinandEn, 2025-02-12:
Hi @tomgolubev37 
The optimization is completely switched off. If I deactivate the MPU, i.e. the program does not require the privileged mode, the display of the variables works. They are therefore not optimized away. Not only the count variable was affected, but also all global variables and the peripheral display.  With the latest version 6.4.0 of the debugger, the problem seems to have been solved. However, I have only tested a few simple programs. I am currently using uVision as a debugger. Everything has always worked there.

Ferdinand

Translated with www.DeepL.com/Translator (free version)