ARM-software/vscode-arm-debugger #14: How to start debugger for STM32H747 Multicore

FerdinandEn, 2024-11-04:
Type: Bug Report

**Describe the bug**

- Windows 11 23H2
- VSC 1.95.1
- Keil Studio 1.18.3
- Arm-Debugger 6.2.1 / 1.2.2
- STM32H747 Discovery Board

I have tried to debug a STM32H747 multicore. As soon as I try to flash or to start the debugger, I get the following error message:

5 NOT_FOUND failed to get resolved hardware ErrorMessage: device=STM32H747XIHx has many processors, specify which one to use

**To Reproduce**
Steps to reproduce the behavior:
1. Select CMSIS
2. Click on Triangle

**Expected behavior**

Flashing or start Debugger

**Code sample and logs**
- See attached example
- `tasks.json`

````
{
    "version": "2.0.0",
    "tasks": [
        {
            "type": "arm-debugger.flash",
            "serialNumber": "${command:device-manager.getSerialNumber}",
            "programs": "${command:cmsis-csolution.getBinaryFiles}",
            "cmsisPack": "${command:cmsis-csolution.getTargetPack}",
            "deviceName": "${command:cmsis-csolution.getDeviceName}",
            // "processorName": "${command:cmsis-csolution.getProcessorName}",
            "processorName": "cm7",
            "problemMatcher": [],
            "label": "Flash Device CM7"
        }
    ]
}
````

How can I specify which processor I need to address? I have already tried the following combinations, for example: "cm7", "CM7", "cm7", "CM7", "0", "1"

shuky-shukrun, 2024-11-13:
Hi,

Specifying `cmsisPack` solved the same issue for me:
`
            "cmsisPack": "Keil::STM32H7xx_DFP@4.0.0",
`

FerdinandEn, 2024-11-14:
@shuky-shukrun 

Thanks for the tip, now I can continue testing.