ARM-software/vscode-arm-debugger #15: The License Manager Utility or ARMLM Cannot Recognize ARMLMD Server License

huiwangbcm, 2025-02-05:
Type: Bug Report

**Describe the bug**
- OS and Version: macOS Sonoma 14.4
- VS Code Version: 1.96.4
- Extension Version: 1.2.2
- Target Device: STM32U5xx

**To Reproduce**
Steps to reproduce the behavior:
1. Add the following in task.json
  {
   "type": "arm-debugger.flash",
   "serialNumber": "${command:device-manager.getSerialNumber}",
   "program": "${command:arm-debugger.getApplicationFile}",
   "cmsisPack": "${command:cmsis-csolution.getTargetPack}",
   "deviceName": "${command:cmsis-csolution.getDeviceName}",
   "processorName": "${command:cmsis-csolution.getProcessorName}",
   "problemMatcher": [],
   "label": "arm-debugger.flash: Flash Device",
   "probe": "ST-Link",
   "programs": [],
   "flms": [
    {
     "path": "CMSIS/Flash/STM32U5xx_1M_0C00.FLM",
     "ignore": true
    }
   ]
  },
2. Execute task "arm-debugger.flash"
3. See error
Executing task: arm-debugger.flash: Flash Device 

ERROR(CLI196): 
               ! Unable to acquire license for debug session. The licensing environment appears to be unconfigured. Activate using either 'armlm' or Arm Development Studio IDE and complete the Product Setup Wizard.
           ! Licensing Manager failed to initialise with the error: Failed to locate the license mapping file in root directory with default toolkit.

Flash failed

 *  The terminal process terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

arronhartley-arm, 2025-02-06:
Hi @huiwangbcm . What license technology are you expecting to use? 'Flex' licensing is not supported in this version of the debugger. Conversely, User-based licensing does not require any environmental variables set, nor reads from .flexlmrc. The armlm tool is used to configure Use-based licensing.

huiwangbcm, 2025-02-07:
Hi @arronhartley-arm , thank you for your reply! I have a flexnet license in the format of <port>@<hostname>. I have been using it by adding it in the flex license under Keil MDK license management on Windows PC. For macOS, I have been exporting it as ARMLMD_LICENSE_FILE=<port>@<hostname> and armclang seem to work fine as well,

 - ARMLMD_LICENSE_FILE: '<example port>@<example hostname'
 - LM_LICENSE_FILE: unset
 - ARM_PRODUCT_DEF: unset
 - ARM_PRODUCT_PATH: unset
 - ARM_TOOL_VARIANT: unset
 - Product location: /Users/exampleusername/.vcpkg/artifacts/2139c4c6/compilers.arm.armclang/6.21.0/sw/mappings
 - Toolchain location: /Users/exampleusername/.vcpkg/artifacts/2139c4c6/compilers.arm.armclang/6.21.0/bin
 - Selected tool variant: product
armclang: note: 
 - Checkout feature: compiler5
 - Feature version: 5.0202310
Product: Arm Compiler for Embedded 6.21 Professional
Component: Arm Compiler for Embedded 6.21
Tool: armclang [5ec1f900]

Do you know if arm debugger under the vscode keil studio pack would also be able to recognize this kind of flexnet license? The same flexnet license added to the MDK in windows will enable Uvision for debug session as well as flash download.

Thanks!
Best,
Hui