ARM-software/vscode-arm-debugger #16: Cannot download to Cortex-M23 device

jamison-chiu, 2025-03-03:
- Windows 10 Enterprise 22H2

- VSCode 1.97.2

- Arm-Debugger 1.4.4

- Arm-Embedded-Debugger 1.2.2

I have a NuMaker-M252KG board and I develop my project in VSCode with Arm CMSIS Solution extensions.
I can download my code with Arm-Embedded-Debugger successfully.
But I cannot download code with Arm-Debugger, it reports error when connecting to device as below.
How can I solve this error ? Thanks.

```
Copyright (C) 2008-2025 Arm Limited (or its affiliates). All rights reserved.

Starting Arm Debugger server. Waiting for remote connection on port XXXXXXXX/10.103.61.166:55000

(Client CLI connected from 127.0.0.1:55847)

Initializing [DONE] Product license
Initializing [DONE] CMSIS Packages
Initializing [DONE] Initializing database
Initializing [DONE] 
Initializing [DONE] Connecting
(ERROR:CLI0037): Debug session launch failed
(ERROR:SRV37): "Cortex-M23" does not match any device; available devices are:
   1 ARMCS-DP
   2 CSMEMAP_0
   3 Cortex-M4
(Client CLI disconnected from 127.0.0.1:55847)
Terminating Arm Debugger server on OANBE03685/10.103.61.166:55000
Flash failed
```

My tasks.json
```
{
	"version": "2.0.0",
	"tasks": [
		{
			"type": "arm-debugger.flash",
			"serialNumber": "${command:device-manager.getSerialNumber}",
			"program": "${command:arm-debugger.getApplicationFile}",
			"cmsisPack": "${command:cmsis-csolution.getTargetPack}",
			"deviceName": "${command:cmsis-csolution.getDeviceName}",
			"processorName": "${command:cmsis-csolution.getProcessorName}",
			"problemMatcher": [],
			"label": "arm-debugger.flash: Flash Device"
		}
	]
}
```

arronhartley-arm, 2025-03-06:
Hi @jamison-chiu, thanks for reporting. During the initial connection process, the Arm Debugger attempts to construct some metadata describing the platform, but it looks like something has gone wrong during this process.

You can try work around this by modifying the cached metadata file that Arm Debugger generates.

On Windows, this is in `%LOCALAPPDATA%/Arm/armdbg/sdf`
In MacOS/Linux, this is in `~/.cache/Arm/armdbg/sdf`

Open the .sdf file in this location with a text editor and change the line that looks something like:

`<device name="Cortex-M4" type="Cortex-M4">`

to:

`<device name="Cortex-M23" type="V8M-Generic">`

We'll look into why this is generating the wrong metadata.

jamison-chiu, 2025-03-07:
Hi Arron,
Thank you for providing the information.
The cached metadata file path on my PC is located at _%LOCALAPPDATA%\Temp\xxx.sdf_.
After modifying the device name to match my target board, I was able to download the code successfully.

I am working with four target boards: Cortex-M0, M23, M4, and M55. 
However, when switching between different boards, such as from M4 to M23, the cached metadata file does not regenerate.

arronhartley-arm, 2025-03-07:
Hi @jamison-chiu . I'm glad you were able to download code successfully.

I now understand the issue. Nuvoton boards use a single CMSIS-Pack to describe a whole range of boards ranging from Cortex-M0 to M55, as you're likely using.

The Arm Debugger will generate an SDF file for a single CMSIS-Pack, which in this case doesn't work because each board described in the single CMSIS-Pack is quite different (from a CPU perspective). Arm Debugger requires the SDF to match the target board to load architecture-specific information (v8-M is quite different to v6-M).

I'll raise this issue internally and update you on any progress. I think for now unfortunately you'll need to wipe the cached SDF when you switch between boards.

jamison-chiu, 2025-03-11:
Hi Arron,

Thank you for the information.

We have divided the Nuvoton CMSIS-Pack into multiple pack files, each containing a single architecture. 
This approach has allowed us to generate separate SDF files for each core and successfully download code across Cortex-M0, M23, and M4.

However, we are encountering an issue with the Cortex-M55: the generated file lacks **_CORESIGHT_AP_ADDRESS_** information.
The M55 pack file describes this M55 device using _accessportV2_ with ___apid="0"_ and _address="0x00000000"_.
Do you have any suggestions on how to resolve this issue?

```
<device Dname="M55M1">
          <accessportV2 __apid="0" address="0x00000000"/>
```

```
<platform>
        <scanchain>
            <dap irLength="4" name="ARMCS-DP" type="ARMCS-DP">
                <device_info_items>
                    <device_info_item name="EXCLUDE_FROM_DTSL">False</device_info_item>
                    <device_info_item name="DP_VERSION">DPv2</device_info_item>
                </device_info_items>
                <device name="CSMEMAP_0" type="CSMEMAP">
                    <config_items>
                        <config_item name="CORESIGHT_AP_ADDRESS">0x00000000</config_item>
                                           ^^^^^^^^^^^^^^^^^^^^ I need to add this config_item manually.
                        <config_item name="CORESIGHT_AP_INDEX">0</config_item>
                        <config_item name="version">APv2</config_item>
                    </config_items>
                    <device_info_items>
                        <device_info_item name="EXCLUDE_FROM_DTSL">False</device_info_item>
                        <device_info_item name="AP_TYPE">AHB-AP</device_info_item>
                        <device_info_item name="ROM_TABLE_BASE_ADDRESS"/>
                        <device_info_item name="AP_VERSION">APv1</device_info_item>
                    </device_info_items>
                </device>
                <device name="Cortex-M55" type="Cortex-M55">
                    <config_items>
                        <config_item name="CORESIGHT_AP_INDEX">0</config_item>
                    </config_items>
                    <device_info_items>
                        <device_info_item name="EXCLUDE_FROM_DTSL">False</device_info_item>
                    </device_info_items>
                </device>
            </dap>
        </scanchain>
    </platform>
```

arronhartley-arm, 2025-03-17:
Hi again @jamison-chiu . We had a bug when generating CMSIS-Pack data from ADIv6 devices. This has been fixed and will be present in Arm Debugger 6.5.0, scheduled for release at the end of this month. I'll drop a comment on here when the release has been pushed.

ccli0, 2025-04-01:
Hi Arron,

I have a NuMaker-M2354 board and I develop my project in VSCode with Arm CMSIS Solution extensions.
I cannot download code with Arm-Debugger, it reports error when connecting to device as below.
How can I solve this error ? Thanks.

```
Copyright (C) 2008-2025 Arm Limited (or its affiliates). All rights reserved.

Starting Arm Debugger server. Waiting for remote connection on port OANBE04775/10.103.40.84:55000

(Client CLI connected from 127.0.0.1:52158)

Initializing [DONE] Product license
Initializing [DONE] CMSIS Packages
Initializing [DONE] Initializing database
Initializing [DONE] 
Initializing [                =========         ] Initialising Debugger
Stopping running target Cortex-M23 on connection
Initializing [                  =========       ] Initialising Debugger
Execution stopped in Secure Privileged Thread mode due to a breakpoint or watchpoint: 0x0003B7BC
Initializing [DONE] Initialising Debugger
0x0003B7BC   MOVS     r0,r0
Connected to stopped target Cortex-M23
Starting flash programming...
Target has been reset
Mapping segment 0x00000000 ~ 0x00001C70 (size 0x1C70)
Execution stopped in Secure Privileged Thread mode at 0x0003B7BC
0x0003B7BC   MOVS     r0,r0
Disconnected from stopped target Cortex-M23
(ERROR:CLI0037): Debug session launch failed
Flash programming failed: General error on memory or register access.
(Client CLI disconnected from 127.0.0.1:52158)
Terminating Arm Debugger server on OANBE04775/10.103.40.84:55000
Flash failed
```

Nuvoton.NuMicroM23_DFP.pdsc file
```
     <!-- *************************  Subfamily 'M2354'  ***************************** -->
      <subFamily DsubFamily="M2354">
        <processor  Dcore="Cortex-M23" Dfpu="NO_FPU" Dmpu="MPU" Dtz="TZ" Dendian="Configurable" Dclock="96000000"/>
        <compile    header="Device/M2354/Include/M2354.h"/>
        <description>The NuMicro M2354 series is a 32-bit microcontroller with embedded ARM Cortex-M23 core.</description>
        <!-- *************************  Device 'M2354KJFAE'  ***************************** -->
        <device Dname="M2354KJFAE">
          <debug      svd   ="SVD/Nuvoton/M2354.svd"/>
          <memory     id="IROM1"    start="0x00000000"  size="0x100000"    startup="1"   default="1"/>
          <memory     id="IROM2"    start="0x10000000"  size="0x100000"    startup="0"   default="0"/>
          <memory     id="IRAM1"    start="0x20000000"  size="0x40000"     init   ="0"   default="1"/>
          <memory     id="IRAM2"    start="0x30000000"  size="0x40000"     init   ="0"   default="0"/>
          <algorithm  name  ="Flash/M2354_AP_1M.FLM"      start="0x00000000"  size="0x100000"  RAMstart="0x20000000"  RAMsize="0x4000"  default="1"/>
          <algorithm  name  ="Flash/M2354_LD_16.FLM"      start="0x00100000"  size="0x4000"    RAMstart="0x20000000"  RAMsize="0x4000"  default="1"/>
          <algorithm  name  ="Flash/M2354_NS.FLM"         start="0x10000000"  size="0x100000"  RAMstart="0x30000000"  RAMsize="0x4000"  default="1"/>
          <feature type="IOs"       n="106"/>
          <feature type="Timer"     n="4"   m="32"/>
          <feature type="UART"      n="6"/>
          <feature type="SPI"       n="4"/>
          <feature type="I2C"       n="3"/>
          <feature type="USBD"      n="1"/>
          <feature type="LIN"       n="2"/>
          <feature type="CAN"       n="1"/>
          <feature type="I2S"       n="1"/>
          <feature type="PWM"       n="12"  m="16"/>
          <feature type="ADC"       n="16"  m="12"/>
          <feature type="DAC"       n="2"   m="12"/>
          <feature type="QFP"       n="128"/>
        </device>
```

[... 94 lines cut ...]
                    <config_item name="RddiProbeConfig">PrioritizeV2=True</config_item>
                </connection_config>
            </debug_vehicle_connection>
            <debug_vehicle_connection>
                <connection_config type="FTDI MPSSE JTAG">
                    <config_item name="RddiProbeConfig">GPIOL0=Low</config_item>
                    <config_item name="RddiProbeConfig">GPIOL1=High</config_item>
                    <config_item name="RddiProbeConfig">GPIOL2=High</config_item>
                    <config_item name="RddiProbeConfig">GPIOL3=Input</config_item>
                    <config_item name="RddiProbeConfig">GPIOH0=TRST</config_item>
                    <config_item name="RddiProbeConfig">GPIOH1=SRST</config_item>
                    <config_item name="RddiProbeConfig">GPIOH2=High</config_item>
                    <config_item name="RddiProbeConfig">GPIOH3=High</config_item>
                    <config_item name="RddiProbeConfig">GPIOH4=High</config_item>
                    <config_item name="RddiProbeConfig">GPIOH5=High</config_item>
                    <config_item name="RddiProbeConfig">GPIOH6=High</config_item>
                    <config_item name="RddiProbeConfig">GPIOH7=High</config_item>
                    <config_item name="RddiProbeConfig">SRST_Invert=True</config_item>
                    <config_item name="RddiProbeConfig">TRST_Invert=False</config_item>
                </connection_config>
            </debug_vehicle_connection>
        </debug_vehicle_connections>
    </extension>
</system_description>
```

ccli0, 2025-04-18:
Hi @jamiebird-arm,

We are already using version 6.5.0, but the issue still persists.
The issue has been resolved for version 6.5.0 on the M55, but it still remains unresolved for the M2354 and M2351.

ccli0, 2025-05-20:
Hi @jamiebird-arm,

any update?

We are using another M23 device without TrustZone, and we are encountering a similar situation.

```
Arm Debugger version 6.5.0+5
Copyright (C) 2008-2025 Arm Limited (or its affiliates). All rights reserved.

Starting Arm Debugger server. Waiting for remote connection on port OANBE04775/10.103.63.147:55000

(Client CLI connected from 127.0.0.1:61206)

Initializing [DONE] Product License
Initializing [DONE] CMSIS Packages
Initializing [DONE] Initializing Database
Initializing [DONE] 
Stopping running target Cortex-M23 on connection
Execution stopped in Non-secure Handler mode due to a breakpoint or watchpoint: 0xEFFFFFFE
Initializing [DONE] Initializing Debugger
0xEFFFFFFE   MOVS     r0,r0
Connected to stopped target Cortex-M23
Starting flash programming...
Target has been reset
Mapping segment 0x00000000 ~ 0x00000B30 (size 0xB30)
Programming 2864 bytes to 0... [DONE] Flash Progress: Writing completed

Execution stopped in Non-secure Handler mode at 0xEFFFFFFE

0xEFFFFFFE   MOVS     r0,r0

Disconnected from stopped target Cortex-M23
Flash programming failed: Failed to verify at address 0x00000b30
(ERROR:CLI0037): Debug session launch failed
(Client CLI disconnected from 127.0.0.1:61206)
Terminating Arm Debugger server on OANBE04775/10.103.63.147:55000
Flash succeeded *  Terminal will be reused by tasks, press any key to close it. 
```