ARM-software/vscode-arm-debugger #17: FVP fails to launch when using launch configuration with "programs" and "${command:cmsis-csolution.getBinaryFiles}"

VladimirUmek, 2025-03-27:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11
- VS Code Version: 1.98.2
- Extension Version: Arm debugger 1.4.4 and Arm CMSIS Solution 1.50.0
- Target Device: FVP_MPS3_Corstone_SSE_310

I have secure/non-secure application and two axf images that have to be loaded to FVP model. To do this I'm using launch configuration with _"programs"_ and _"${command:cmsis-csolution.getBinaryFiles}"_.
The launch configuration example:
```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "name": "CM85",
      "type": "arm-debugger.fvp",
      "request": "launch",
      "cdbEntry": "Arm SubSystem FVP::Corstone SSE-310 Ethos-U55 (MPS3)::Bare Metal Debug::Bare Metal Debug::Cortex-M85",
      "fvpParameters": "\"${workspaceFolder}\"/Target/CM85/fvp_config.txt",
      "debugFrom": "Reset_Handler",
      "programs": [
        "${command:cmsis-csolution.getBinaryFiles}"
      ]
    }
  ]
}
```
When launching the model from the "Run and Debug" view in VS Code by pressing "Start Debugging" button, the model seems to be launched, but the connection is quickly terminated due to the error. The output that I see is (slightly modified to be more condensed):
```
Initializing Arm Debugger...
[model] Iris server started listening to port 7100
[model] Info: FVP_MPS3_Corstone_SSE_310: telnetterminal0: Listening for serial connection on port 5000
[model] Info: FVP_MPS3_Corstone_SSE_310: telnetterminal1: Listening for serial connection on port 5001
[model] Info: FVP_MPS3_Corstone_SSE_310: telnetterminal2: Listening for serial connection on port 5002
[model] Info: FVP_MPS3_Corstone_SSE_310: telnetterminal5: Listening for serial connection on port 5003
[model] Iris server is reported on port 7100
Connected to stopped target Cortex-M85
Illegal char <:> at index 110: c:\D\GitHub\ARM-software\CMSIS-FreeRTOS\CMSIS\RTOS2\FreeRTOS\Examples\out\TZ_Secure\CM85\Debug\TZ_Secure.axf,c:\D\GitHub\ARM-software\CMSIS-FreeRTOS\CMSIS\RTOS2\FreeRTOS\Examples\out\TZ_NonSecure\CM85\Debug\TZ_NonSecure.axf
(ERROR:INT31): An unexpected error has occurred. Please report to your local support representatives.
```

**To Reproduce**
Steps to reproduce the behavior:
Use [CMSIS-FreeRTOS Examples](https://github.com/ARM-software/CMSIS-FreeRTOS/tree/bb1d1243182161ebad0ce0171c59c4351ec98fe9) to reproduce the behavior. Open "Examples" directory with VS Code. The only required change is launch configuration modification as mentioned above.

**Expected behavior**
FVP model should launch with both images loaded.

**Additional context**
Additional problem when working with this particular case is:
For TrustZone application with secure/non-secure images, the images have to be loaded in correct order (when using FVP). Non-secure first and secure image second (due to the automatic entry point recognition). This is also something that has to be taken into account.

StavrosKolliasArm, 2025-08-12:
Hi @VladimirUmek, 

Thanks for raising this issue and apologies for the late reply. 
We would recommend to update our Arm debugger extension to the latest released version 1.6.0 
We think you have to explicitly set each image files in this version.

Here is an example with a configuration that suits this example, please change the reference your_path to the  model_params

```
    {
      "name": "CM85",
      "type": "arm-debugger.configdb",
      "request": "launch",
      "cdbEntry": "Arm SubSystem FVP::Corstone SSE-310 Ethos-U55 (MPS3)::Bare Metal Debug::Bare Metal Debug::Cortex-M85",
      "programs": [
        {
          "path": "${workspaceFolder}/out/TZ_Secure/CM85/Debug/TZ_Secure.elf"
        },
        {
          "path": "${workspaceFolder}/out/TZ_NonSecure/CM85/Debug/TZ_NonSecure.elf"
        }
      ],
      "cdbEntryParams": {
        "model_params": "-f /your_path/CMSIS-FreeRTOS/CMSIS/RTOS2/FreeRTOS/Examples/Target/CM85/fvp_config.txt"
      },
      "runControl": {
        "debugFromSymbol": "main"
      },
      "resetAfterConnect": true
    }
```

Please do let me know if there is anything else. 

Kind regards, 
Stavros Kollias

VladimirUmek, 2025-09-30:
Many thanks for status updates!

May I just ask what is the load order in case of secure/non-secure application? Is there a way to control loading order?

closed 2025-12-17 by VladimirUmek